jareware/chilipie-kiosk #117: Portrait mode problem

klaaskox, 2020-06-03:
I am using display_rotate=3 to rotate the screen. Now the orientation of the screen is correct, however, the dimensions of the viewport/monitor seem too be wrong, see attached image.

[image]

What can I do to make the browser fit the monitor?

klaaskox, 2020-06-03:
I think I found the solution (Raspberry Pi 4). 

disabled vc4-fkms-v3d in `/boot/config.txt`:
`#dtoverlay=vc4-fkms-v3d`

added rotation to `/boot/config.txt`:
`display_hdmi_rotate=3`

closed 2020-06-03 by klaaskox

wujood, 2020-09-23:
You do not need to disable dtoverlay since this might reduce performance (especially for videos). Please look at issue https://github.com/futurice/chilipie-kiosk/issues/92