jareware/chilipie-kiosk #118: How to force a refresh of the web page via CLI

rklueber, 2020-06-07:
Hello,

is there a way to login via SSH and force a reload of the page?

Regards
Ralf

allcoolusernamesaregone, 2020-06-08:
You could use the example of the hourly reload of the website as contained in the crontab of user pi

# Example: Reload webpage every hour
# 0 * * * * xdotool key ctrl+R

rklueber, 2020-06-09:
Hi, thanks 1000 times. Script I added was:

reload.sh
```
#!/bin/bash

DISPLAY=:0 xdotool key ctrl+R
```

works like a charm.

Regards
Ralf

closed 2020-06-09 by rklueber

royatkup, 2024-11-14:
This works indeed (in most cases)!

BUT; it does not clear the cache, so i'm loading a webpage with a .mp4 file where only the video file (.mp4) changes.
Problem is, when we use the command like mentioned above, it keeps showing the same video.
Only after a reboot, or manual restart the chromium-browser it will show the new video.

I already tried the following:

```shell
DISPLAY=:0 xdotool key ctrl+shift+R
``` 

or

```shell
DISPLAY=:0 xdotool key F5
``` 
With both commands you see the page is reloading, but it does not load the new video file.

Anybody have an idea?