jareware/chilipie-kiosk #119: Wifi Connection with USB Dongle

fra-sabba, 2020-06-22:
Hello,
I downloaded and flashed the last image for my Raspberry Pi 2.
I created a wpa_supplicant.conf file offline, I put it in the boot partition and I powered on the Raspi.
I have a TP-LINK Wifi USB dongle (model: TL-WN725N) that is compliant with Raspberry Pi 2 board, I used it many times with Raspbian OS on this type of board.

For the wpa_supplicant file I used the example file and I replace correctly ssid and psk with my values, but I'm not able to connect to my local home network.

The USB dongle is recognized correctly: it is in the list of USB devices ('lsusb' command). Moreover, I can scan and find successfully my home wifi network with the USB dongle along with the command: 

`sudo iwlist wlan0 scan`

But I cannot execute the command (https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md) to reconfigure wlan0 with the wpa_supplicant created previously:

`sudo wpa_cli -i wlan0 reconfigure`

I receive the error:

`Failed to connect to non-global ctrl_iframe: wlan0 error: No such file or directory`

So, I am not connected to the network, but USB dongle seems to work well.

Someone could help me to understand this issue?

Thank you

fra-sabba, 2020-06-22:
Sorry, I resolved the problem with the solution explained here: https://raspberrypi.stackexchange.com/questions/67311/failed-to-connect-to-non-global-ctrl-ifname-when-running-wpa-cli-reconfigure

I added to  /etc/network/interfaces the lines:
```
auto wlan0
iface wlan0 inet manual
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

jareware, 2020-10-08:
OK, thanks for detailing your eventual solution, I'm sure it'll help someone else with the same issue!

closed 2020-10-08 by jareware