jareware/chilipie-kiosk #131: Pi4 GPU Acceleration and Flickering resolve

Nixon666tmp, 2020-10-09:
Hi,

First --> "great job" for Chilipie Kiosk ^^ i love it !!

I have been using Chilippie Kisok for several months on a Raspberri Pi4, and so far have been facing a slow display issue on animated web pages and displaying video.

The problem stemmed from the fact that hardware acceleration is not enabled base and I could not find a post that discusses it. I would therefore like to offer you a solution to enable hardware acceleration on the Candidate 2 release:

1. Config the Raspberri :
```
pi@chilipie-kiosk:~$ sudo -i
pi@chilipie-kiosk:~$ raspi-config
```

Increase the memory allocated to the GPU to 512M
> 7 Advanced Options
> > A3 Memory Split : 512

Activate the "KMS" driver
> 7 Advanced Options
> > A8 GL Driver
> > > G2 GL (Fake KMS)

Extend system partitions
> 7 Advanced Options
> > A1 Expand Filesystem

Finish the configuration and reboot the raspberry
> <Finish>
> >  Reboot : Yes

 2. Install GPU hardware acceleration support libraries :
```
pi@chilipie-kiosk:~$ sudo -i
root@chilipie-kiosk:~# apt-get update
root@chilipie-kiosk:~# apt-get install libgles2-mesa libgles2-mesa-dev xorg-dev
```
 
3. Update basic tools : 
`root@chilipie-kiosk:~# apt-get install matchbox-window-manager xserver-xorg x11-xserver-utils unclutter xinit chromium-browser`
 
Note: At this step if you restart the Raspberry, Chromium generates a Bug after a few seconds resulting in flickering (image bounce between the memory cache and the actual display ... like a strobe effect)
 
4. Resolving the Flickering effect
Edit /home/pi/.xsession

Edit the end of the file as follows:
```
# Start and detach Chromium
# http://peter.sh/experiments/chromium-command-line-switches/
# Note that under matchbox, starting in full-screen without a window size doesn't behav$
chromium-browser \
  --start-fullscreen \
  --window-position=9000,9000 \
  --disable-infobars \
  --check-for-update-interval=1 --simulate-critical-update \
  &
matchbox-window-manager -use_titlebar no
# See https://github.com/futurice/chilipie-kiosk/issues/99#issuecomment-597119842 for t$
 
# Hide Chromium while it's starting/loading the page
#wid=`xdotool search --sync --onlyvisible --class chromium`
#xdotool windowunmap $wid
#sleep 10 # give the web page time to load
#xdotool windowmap $wid
 
# Finally, switch process to our window manager
#exec matchbox-window-manager -use_titlebar no
```

5. Reboot Raspberry pi

Now if you check browser parameters GPU accelerations should be activate. Please, Let me know if it work

jareware, 2021-02-24:
Awesome work figuring this out and documenting it so well!

Let's try and include this in the docs at least, so others can do the same.

I'm not sure if this can be included in the base distribution as-is, since it has to work across a wide range of Raspberry Pi models, not just the 4. 🤔

adoconnection, 2021-03-05:
this also works for Pi3B+
OMG css animations got so much faster and smoother!!!

the only difference is 3B+ model allows only 256mb of GPU memory

Nixon666tmp, 2021-03-25:
Good news ^^ thank for your feedback

federicotravaini, 2021-04-09:
I don't have option 7 in the raspi-config menu. How can I open it?

Nixon666tmp, 2021-04-09:
Hi,

Do you open raspi-config with sudo ?

Maybe you should find more answers here : [https://github.com/asb/raspi-config/](https://github.com/asb/raspi-config/)