jareware/chilipie-kiosk #132: Delete Browser Cache how to?

davem50, 2020-10-14:
Hello,
I am new here. I hope my english is good enough to explain my issue.
I have running 12 raspberry pi using chilipi Kiosk. 
Now I must delete the Browser Cache from each rpi after a system update. 
Can you help me how I have to do this?
Must I stop the browser first? 
Can I delete the cache in the command line?

Because when I leave the full screen mode, there is no option menue like in the normal desktop Browsers on a workstation. 

I hope you can help me.

Many thanks. 

Greetings from Germany, 
davem50

markovchainz, 2020-11-07:
@davem50  This should help you:
```
sudo rm -r ~/.cache/chromium
sudo rm -r ~/.config/chromium/Default
```

Just be aware that the second line sudo rm -r ~/.config/chromium/Default will remove any loaded extensions (i.e. virtual keyboards), local storage data and settings saved.

jareware, 2021-02-24:
Hope this got sorted, and thanks @markovchainz for providing assistance!

closed 2021-02-24 by jareware