jareware/chilipie-kiosk #135: PROJECT ABANDONNED?

codeclinic, 2020-11-14:
It appears that there has been no activity on this project for 9+ months now.

Futurice has said previously that it's not intended for users to update the image via apt, and should only update via bumping to the latest Chilipie-Kiosk version. This means that systems would be 9+ months out of date by now, which is not only unadvisable it also poses potential security issues.

Therefore.... could Futurice confirm whether or not the project is still being developed? Or, has it now been abandonned?

Eitherway, thanks for all the hard work and time put into this project.

ThisIsTenou, 2020-11-14:
This is imo, especially considering the latest security issues with chromium, a pretty important question. Especially for people eventually using this in a business environment (although I absolutly get that this isn't a commercial software where any kind of support is garantueed - it's still a project from someone doing this for free in his free time, much kudos for that) it'd be great to have an updated image.

markovchainz, 2020-11-14:
If there’s a need for help, I am willing to help out. I am considering forking and doing a major overhaul but I would prefer to contribute here instead.

codeclinic, 2020-11-14:
@markovchainz It's been 9months... Sooooo... 

- [ ] Fork it!

markovchainz, 2020-11-15:
Done - [https://github.com/markovchainz/chilipie-kiosk](url)

codeclinic, 2020-11-15:
> 
> 
> Done - [https://github.com/markovchainz/chilipie-kiosk](url)

Urm.... nope. That link doesn't work.

jareware, 2021-02-24:
Hey folks, maintainer here!

As is often the case with free/OSS projects, the maintainers have day jobs, families and other distractions which sometimes take precedence. Thanks all for your kind words in understanding that!

And with that out of the way, the project is definitely not abandoned, though admittedly it's been too long since the last release. Let's fix that.

@codeclinic 

> it's not intended for users to update the image via apt, and should only update via bumping to the latest Chilipie-Kiosk version

This is definitely true, though of course nothing prevents you from doing it, and nothing has been done to make it any more difficult than it is by default. This is, after all, just a pre-curated list of packages and configuration tweaks on top of stock Raspbian.

@ThisIsTenou 

> security issues with chromium, a pretty important question

I don't want to downplay any security issues, but also the intended use case for this project is for web kiosks or wall displays, where you control both the physical setting and the website being shown. If the Grafana dashboard you render serves malware, then that's not great, but also probably quite unlikely. Oftentimes they might even be on the same LAN, with no access to the internet.

I know there's folks out there running this as part of a web kiosk to which untrusted users might have physical access to, and that's fine, but also (as covered in other discussions here) that's a use case that requires a huge amount of hardening on every level that we're not aiming to - or claiming to - take on.

Any patches toward this will of course be warmly welcomed! 🙂 

Like I said, I definitely don't want to downplay security, but the biggest reason why you need to keep your browsers up to date on your general purpose computing devices is that you use them to execute new, untrusted code from strangers around the internet every day. That is decidedly not the use case for this project.

@markovchainz 

> If there’s a need for help, I am willing to help out

The two biggest time sinks of maintenance for this project are:

1. Updating the base image as new Raspbian releases come out, and updating [the image build script](https://github.com/futurice/chilipie-kiosk/blob/master/docs/image-setup.sh) and any packages & config accordingly, so that it produces a working image.
2. Testing that image by running it on your wall display for a few days, preferably weeks, so that you know that it keeps working if left alone.

This is a lot of work, but also the whole idea behind this project - doing that configuration & testing work for you, so that when you install this image somewhere, it Just Works. ™️ 

Any help in doing either of those would be very, very welcome.

closed 2021-02-24 by jareware

jareware, 2021-02-28:
Speaking of, there's now [a new release candidate image](https://github.com/futurice/chilipie-kiosk/releases/tag/v3.0.0-rc3) with updated everything.

Promoting that to a stable release is now only a question of running it "in production" somewhere, and reporting any issues you find.

**Any help with this will be much appreciated!**