jareware/chilipie-kiosk #140: Problem with rotate display and touchscreen for kiosk

Floranixena, 2021-02-06:
Hi, I have to build a kiosk to sell tickets and i bought the display (ilitek 2511) 21.5 inch. 
i'm configuring it with chilipie kiosk, but i have several problems with the touchscreen and the rotation of the display and touchscreen. (HDMI 1 port of raspberry pi4 B)
how can i best configure with chilipie?

the problem is that with raspberry pi4, with chilipie 3.0, I can't rotate both the display and the touchscreen vertically in /boot/config.txt

i don't know how i can fix it.
do you need a driver?

-------------------------------
I try this but it did not work

Rotate the display direction of the Raspberry Pi screen 0~270 degrees (touch screen and HDMI display)
1. Switch to root

2. Edit /boot/config.txt, according to the angle you want to rotate
For the touch screen, add the following:

lcd_rotate = 1 //rotate 90 degrees

For HDMI display output, add:
If you just want to rotate the angle of the HDMI output display (without touch screen)
You can add:

display_rotate = 1

3. reboot
After restarting, the touch screen driver and display will not rotate.
--------------------------------
can someone give me some advice?
thanks

biasync, 2021-02-18:
I've been following this one. Works fine for me.
https://www.waveshare.com/w/upload/8/82/5inch_HDMI_LCD_%28H%29_User_Manual_EN.pdf

jareware, 2021-02-24:
@Floranixena did that solve your issue?

Also thanks @biasync for helping with support!

mirodrigom, 2021-02-25:
Same happends to me in **RPi3**. Any solution?
When i removed: **dtoverlay=vc4-fkms-v3d** from /boot/config.txt and add **display_rotate=1** works fine, but chromium-browser does not show up, neither splashscreen.
Thank you.

Floranixena, 2021-03-04:
i try, thanks.
I'll let you know
thanks

Floranixena, 2021-03-13:
@biasync I have resolved with your solution, thanks for everything.