jareware/chilipie-kiosk #142: Screensaver

soosterwaal, 2021-02-15:
I want to use a screensaver so that the kiosk goes into screensaving after x minutes, and reactivates on touch.

Can anybody help me with achieving this?

jareware, 2021-02-24:
The screensaver has been explicitly disabled for the image, because it's meant to be used as a wall display, and most folks don't want that.

I think you should be able to re-activate it by booting into graphical mode (via the `raspi-config` utility for example) and turning it back on.

Let us know how that goes.

blueicehaller, 2021-02-25:
In #115 I described an way how to achieve this.

pepsonEL, 2021-03-02:
Did you resolved problem .....? I need to turn off/suspend display after some minutes....to save energy because i use kios mode to panel fo my Home Automation.... and i want to turn off display after 5 min and again turn on when touch screen(display is with touch screen). Any solution ?

blueicehaller, 2021-03-02:
Maybe my Issue/Guide #115 is relevant for you?
I configured it on an 27" Touchscreen which goes into sleep after x minutes (LED changes from blue to yellow) and consumes 0.5 W instead of 36 W.

fdarcy, 2021-03-07:
Well which layer or application has the responsibility to provide and manage the screensaver functionality? I would say its the web application, and not chromium. So depending on how you have setup the environment, here's an example: https://www.npmjs.com/package/react-idle-timer

victormota-al, 2022-11-29:
Has anyone managed to turn off the monitor?