jareware/chilipie-kiosk #146: Chromium follows redirects and then stores wrong URL in session

unnu, 2021-04-08:
I'm not sure that this can be fixed by default, and maybe is a wanted behaviour for others. I open a webpage and understand, that it get's reloaded periodically. The page I open has a weired error management, because, when an server side error happens, it **redirects** to an error page. Chromium is then happily picking up that error page and displaying it instead of the page I wanted.

What I did is, that I added the URL of the page to the .xsession file when chromium is started. So the problem is solved every night, when the raspberry restarts.

What can I do to fix the problem shortly after it happens? let's say every five minutes. I somehow would need chromium to load a pre defined URL every 5 minutes instead of using the **last** URL that was in the location bar.

jareware, 2021-04-23:
Well, there's probably some _correct way_ of doing this, but in the meanwhile, there's also the _hacky way_, which involves using `xdotool` to control Chromium. 🙂 You can find examples to get you started in the [default crontab config](https://github.com/futurice/chilipie-kiosk/blob/master/home/crontab.example#L27).

unnu, 2021-04-23:
So I would basically do?
```
*/5 * * * * xdotool key ctrl+l && xdotool type "https://example.com/foo/bar" && xdotool key Return
```