jareware/chilipie-kiosk #151: Showing blank screen

uplRahulC, 2021-05-29:
After flashing the OS in sd card, when putting it in a Raspberry pi showing blank screen,

uplabhijit, 2021-05-29:
Facing same problem after flashing SD card using the method mentioned in the document. Tried to flash using:

- rip-imager, an option well supported for raspberry pi image flashing
- Etcher
- 'sudo dd bs=1m if=chilipie-kiosk-vX.Y.Z.img of=/dev/rdisk2'

All the above option completes the flashing process but ends up with SD card in unknown file format. As the disk is not read, it's not booting up.

As to confirm the SD card has no problem, I have tried flashing with Raspberry Pi OS (32-bit) and it worked without any trouble.

jareware, 2021-06-16:
Don't suppose you're using Windows? Regardless, did you check out all the solutions in #89, maybe one of them could work for you?

uplabhijit, 2021-06-16:
Ubuntu 20.04 we used and not windows. I will try the suggestion mentioned in #89 and update.