jareware/chilipie-kiosk #153: HTTPS  access problem

qingshuicn, 2021-07-14:
如果是https 没有证书 一周后不会自动跳过，如果解决这个问题？

jareware, 2021-07-22:
Could you elaborate a bit, please?

qingshuicn, 2021-07-26:
> Could you elaborate a bit, please?

Insecure https link, the restart is normal on the first day, and the interface that requires manual intervention will appear on the seventh day
[image]

jareware, 2021-08-09:
I don't think I've ever seen this happen... 😳 Has anyone else hit this?

SnorreSelmer, 2021-09-10:
Only time we've had issues with SSL certificates was when our Raspberries had an incorrect local time. Our network is very strict, so none of the public NTP servers work. Once we fixed access to a local NTP server, everything worked fine again. HTTPS connections can't be created uf the host and client clocks disagree by much more than a few minutes, and will cough up errors. Not sure if that's the problem here though.