jareware/chilipie-kiosk #159: Can the new version be used with a 3B+?

sborn, 2021-11-29:
I have 12 of these running at our non-profit to display our client-staff scheduling. This is done using sheets, which is no longer supported by Jessie's Chromium. Can the new Rpi 4 version image work with 3B+?

jareware, 2021-12-28:
The 4-versions should work with 3 as well. Did you give it a go?

sborn, 2022-03-29:
I have now! I've been struggling, though. The biggest issues are dealing with screen resolution because you can't mess with desktop from raspi-config. Also, F11 seems to do nothing. Do you have easy tips on getting the screen to only show 720 x 480?

jareware, 2022-03-29:
No specific tips on the resolution, but anything you find on Google about Raspbian and resolution should equally apply here!

I've also noticed the F11 thing. Honestly thought it was just my keyboard that's broken. 😄  Please report back if you're able to sort that out!

sborn, 2022-03-29:
> No specific tips on the resolution, but anything you find on Google about Raspbian and resolution should equally apply here!

I did a workaround. I have a weird special use. Things are up and running. Thank you!
 
> I've also noticed the F11 thing. Honestly thought it was just my keyboard that's broken. 😄 Please report back if you're able to sort that out!

Yeah, I use a Pi400 to set things up, and it doesn't even _have_ an F11 key (???). So then when I deployed, I was surprised to find that it didn't do anything anyway.

Is that true of PiOS desktop Chromium?

jareware, 2022-04-04:
> Is that true of PiOS desktop Chromium?

At least I've seen it happen on mine. 🤨

henroFall, 2022-10-12:
I right click and choose to exit full screen, cause ya, F11 no go and that's before I start using VNC to control it