jareware/chilipie-kiosk #161: start.elf: is not compatible

federicotravaini, 2021-12-27:
I think the image needs an update because with the latest board updates it doesn't work anymore if you flash in on an SD card.

Any chance we will get this update?

[image]

jareware, 2021-12-28:
Thanks for bringing this up.

Indeed a new build would be long overdue.

jareware, 2021-12-28:
There's now [a new RC](https://github.com/jareware/chilipie-kiosk/releases/tag/v3.0.0-rc4) with everything updated to the latest versions.

If you can help test it and report back, it would be much appreciated, and would be happy to mark that as a "stable" release for a change!

federicotravaini, 2021-12-29:
That's awesome, will try in the next few days and let you know.

Thank you very very much!

libbierose86, 2022-03-26:
Hi, do you have an update for this? issue, I tried the latest release but still see the issue

federicotravaini, 2022-03-28:
> Hi, do you have an update for this? issue, I tried the latest release but still see the issue

No, there's a new build but I get the same error.
Unfortunately other didn't manage to replicate the issue.

If you get the same error as mine please reply to this: https://github.com/jareware/chilipie-kiosk/issues/164