jareware/chilipie-kiosk #164: 🚀 New release! Help requested! 🙏

jareware, 2021-12-30:
Hello `chilipie-kiosk` community!

After a bit of a hiatus, there's [a new image](https://github.com/jareware/chilipie-kiosk/releases/tag/v3.0.0-rc4) available for download. 🥳

Building - and especially testing - new images takes a not-insignificant amount of effort. And that's somewhat by design: I know a lot of folks like this image specifically because it has been well-tested, and everything Just Works™️ when you first boot it up. There's great ideas for speeding up the build part (#163), but the biggest reason why new releases have been few and far between recently has been that I don't want to make releases without also making sure they work well. It's not enough to just check that the image boots up on a few different RasPi models (all of which I don't even own anymore), but I would also like to ensure main features like display control scripts, preventing screensaver/sleep, automatic crash-recovery etc still work.

**I would love a bit of help from the community for this testing!**

In practice, if you want to help, please go download the [latest Release Candidate image](https://github.com/jareware/chilipie-kiosk/releases/tag/v3.0.0-rc4), try it out on your RasPi, check that most things work, and report back here.

I'll make sure to also thank you in release notes. 🙇 

Let's see if we can make new releases more frequent together!

federicotravaini, 2021-12-30:
Hello,
first of all thank you very much for updating this project which I found to be very useful.

I just installed the new RC but I get the same errore I was getting with the previous version: start.elf is not compatible

[image]

jareware, 2021-12-30:
Interesting... personally I've never seen that error before, and the new image flashed onto my RasPi 4 (and a 2 I had laying around) without issue.

Are you able to flash the latest vanilla Raspbian onto that card/pi?

federicotravaini, 2021-12-30:
Yeah sure, the lastest version of raspian runs just fine.
Even if I use an SD with chillipie-kiosk updated in a working raspberry, it runs.
The issue is with a fresh new installation.

I'll try to be more detailed:

I have 5 pi 4 running chillipie-kiosk. I update all of them regularly. Everything works, expect if i take a new SD and flash chillipie on it: i get the error above. If i put in the same raspberry an SD with the updated chillipie, it runs smoothly.

Even if I clone an SD with the updated chillipie, it runs.

I repeat that when I say "updated chillipie" I mean chillipie already installed on a raspberry, updateg with apt update/upgrade

jareware, 2021-12-30:
Hmm. What do you use for flashing the image onto the SD card?

jareware, 2021-12-30:
(Accidental close, never mind)

federicotravaini, 2021-12-30:
> Hmm. What do you use for flashing the image onto the SD card?

Balena Etcher

jareware, 2021-12-31:
Let's see if someone else can confirm this.

Like I said, I tried flashing this (with the macOS Raspberry Pi Imager app) to both a Pi 4 and a Pi 2, and both booted without issue.

I don't doubt you, just trying to say I can't reproduce this.

federicotravaini, 2021-12-31:
Try this

- install raspbian and full upgrade it
- try again to flash and run chillipie

federicotravaini, 2022-02-23:
Hi, no news about this new release?
Might be a good idea to move to project to docker?

jareware, 2022-03-02:
Sorry, been super busy at the day job. The new RC is still there, and should work for most cases.

Haven't had time to try to reproduce yours. Let us know if you can figure it out.

wrt. docker, sure, but it'll be a LOT of work. Happy to coordinate that work if someone wants to undertake it.

charlie-s, 2022-03-22:
Just flashed onto a R 4 B yesterday successfully, using Balena Etcher on MacOS. Will let a screen run for a few days to test.

Is there a place for community discussion / questions outside of this issue queue? I have some questions for the community but don't want to clutter issue queues, or perhaps I could send you a DM @jareware.

jareware, 2022-03-23:
Thanks for the testing help!

Frankly, if you have a specific topic in mind, feel free to open an issue. You can prefix it with "Discussion:" or something if you'd like. 🙂

congtrol, 2022-06-02:
Thank you very much for this project.  

I have a sound Issue:  

- new RC4 version  couldn't detect  sound device:   
- No HDMI sound device detected, 
- Only headphone is detected.

Tested on Raspberry Pi 3 B+    
- RC2 and RC3 version works well.

pduck27, 2022-10-15:
For me RC4 also misses HDMI sound with a Pi3B+.
The RC3 has sound but the autostart URL will be ignored.

maujabur, 2022-11-02:
> Thank you very much for this project.
> 
> I have a sound Issue:
> 
> * new RC4 version  couldn't detect  sound device:
> * No HDMI sound device detected,
> * Only headphone is detected.
> 
> Tested on Raspberry Pi 3 B+
> 
> * RC2 and RC3 version works well.

Hi! First of all, thanks for all the work!.

Testing here on Pi4, no sound over HDMI, it even doesn´t show on raspiConfig, and headphones are mono

codeclinic, 2023-02-01:
Hi @jareware , Is the project still active? It's been a little quiet for the last 10 or so months....