jareware/chilipie-kiosk #167: Running additional commands in xsession

DaveDischord, 2022-01-14:
I'm creating a web-based timeclock on a touchscreen and wanted to overlay a numerical keypad users can use to punch in.  I have matchbox-keyboard installed and the keypad is set to have everything needed and be the size needed so that if I ran
`DISPLAY=:0 matchbox-keyboard -s 40&` via ssh the keypad displays perfectly.  I have tried having the command ran in crontab  (with and w/o `DISPLAY=:0`)with no luck.  I've also use a script to toggle it on and off.
```
#!/bin/bash
PID="$(pidof matchbox-keyboard)"
if [  "$PID" != ""  ]; then
  kill $PID
else
  DISPLAY=:0 matchbox-keyboard -s 40&
fi
```
My question is, what command and where would I need to put it in xsession in order for the keypad to display after Chromium loads?

DaveDischord, 2022-02-04:
OK.  So I was able to figure this out so I'll paste my solution here for anyone else.
I use this fork: [https://github.com/xlab/matchbox-keyboard](https://github.com/xlab/matchbox-keyboard) because offers a geometry parameter, which ended up being the main problem-solver.  

So as far as getting the keyboard up on boot I put the command
`matchbox-keyboard -g 400x600 &` before the last line
`exec matchbox-window-manager -use_titlebar no`

The 400x600 could be different dependent on screen size, resolution, etc., but  it needs to be there and static or else it'll assume a smaller default size.

closed 2022-02-04 by DaveDischord

jareware, 2022-02-05:
Thanks for documenting your solution!