jareware/chilipie-kiosk #168: Change browser

ascc1973, 2022-01-19:
Hello. I would like to know if I can change Chromium to another browser, such as Firefox or Midori, and if so, how can I do it?
Thank you for your work

jareware, 2022-01-20:
You should be able to install e.g. Firefox just as you normally would onto a Raspbian.

That said, not all of the scripts will then work as before.

Good luck!

closed 2022-01-20 by jareware

ascc1973, 2022-01-20:
"not all of the scripts will then work as before"  That's my problem, what changes should I make to my script for the chilipie-kiosk to work properly with firefox?
As far as I understand with Chromium I can no longer use chrome://flags/#enable-experimental-fullscreen-exit-ui, and the consequence is that it becomes too easy to exit fullscreen, making the kiosk vulnerable to interaction of the users.
Maybe you know how to avoid that behavior in Chromium?
Thank you very much for your work and dedicate your time to answering the doubts of all users.

jareware, 2022-01-25:
I'm sorry but I don't enough about Firefox CLI flags to tell you that off the top of my head.

I'm sure they exist, though, check out the Firefox docs.