jareware/chilipie-kiosk #169: Touch screen calibration

Tommyczech, 2022-01-21:
Hi all! First of all thank you for a great software for Raspberry!

I need to create public kiosk using Elo touch screen. So I downloaded their driver and followed the guide to set it up. Finally it worked but touch function of the screen seems to be mirrored diagonally. If I want to use their calibration app, it needs X Windows. I tried to make it work using command:

`sudo apt-get install xorg-dev`

I installed libXm.so.4 library too but still no success.

I used older version 2.1.0 (newer ones somehow did not work for me) with Raspberry Pi version 1B.

Is there something I am missing? Can anyone please provide a simple step by step guide to set what is needed?

reinierdevl, 2022-01-26:
I don't know about the elo driver but probably there is something similer like this
Where with invert option on or off you can swap direction

sudo nano 99-calibration.conf

Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "Calibration"   "208 3905 288 3910"
        Option  "SwapAxes"      "0"
        Option "EmulateThirdButton" "1"
        Option "EmulateThirdButtonTimeout" "1000"
        Option "EmulateThirdButtonMoveThreshold" "300"
        Option "InvertY" "0"
        Option "SwapXY" "0"
        Option "InvertX" "1"
EndSection

Tommyczech, 2022-01-28:
Thank you for the advice. I have tried to look for somethink similar in my driver directory but I cannot see optins like these. I will probably try to invert screen globaly in system settings. If I find working solution, I will post it here.

reinierdevl, 2022-01-28:
look if you have this directory

/etc/X11/xorg.conf.d

Tommyczech, 2022-01-28:
I could not find directory like that.

I used `sudo nano /boot/config.txt` and wrote `display_hdmi_rotate=2` to invert screen and it works just fine for me.

Now I am trying to hide cursor completely. Is there any config where I could do this?

reinierdevl, 2022-01-28:
unclutter will prevent that

sudo apt-get install -y unclutter

Tommyczech, 2022-02-03:
I have tried this. It seems like unclutter is already included in this version. Pointer disappers after few seconds of inactivity. I would prefer to hide it forever.

closed 2022-02-04 by Tommyczech