jareware/chilipie-kiosk #171: v.3.0.0 vs. v.2.1.0 on a Pi Zero W

ebolisa, 2022-02-03:
Hi,
Has anyone installed the new img on a Pi Zero? I cannot get it to work, it stalls at the boot image. I can ssh in, however.
When I installed v.2.1.0 worked on a first try. I was able to rotate the screen (not able to do it with v.3.0.0) and modify the html boot page to redirect it to my MagicMirror2 server with no issues.
TIA

jareware, 2022-02-05:
Don't have a Zero W at hand for testing, unfortunately.

Please let us know if you can sort it out.

federicotravaini, 2022-02-23:
Might be the same issue I mentioned here?

https://github.com/jareware/chilipie-kiosk/issues/164#issuecomment-1002929821

Surebob, 2022-08-16:
For anyone with a PI Zero, you have to downgrade chromium, this particular version is 88 and tested by me.
 
`wget "http://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/chromium-browser_88.0.4324.187-rpt1_armhf.deb"
wget "http://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/chromium-codecs-ffmpeg-extra_88.0.4324.187-rpt1_armhf.deb"
apt install --no-install-recommends --allow-downgrades --allow-change-held-packages ./chromium-browser_88.0.4324.187-rpt1_armhf.deb ./chromium-codecs-ffmpeg-extra_88.0.4324.187-rpt1_armhf.deb`

you can also hold the packages so they don't update and break things

`apt-mark hold chromium-browser chromium-codecs-ffmpeg-extra`

dont forget to delete downloaded packages.

Cheers!