jareware/chilipie-kiosk #172: Every restart a new tab of chromium is opened

flivolsi76, 2022-02-18:
Every time i start the kiosk a new tab of chromium is opened.
Is there way to close all tabs before everything is started?

flivolsi76, 2022-02-22:
I have fixed it using the property --app

`chromium-browser --kiosk --app=$URL`

kleini, 2022-04-13:
I have the same issue and this massively slows down chromium as I am displaying an overview of a surveillance system.

majdzik84, 2022-12-20:
Have you solved the problem with tabs?

majdzik84, 2022-12-20:
> > Have you solved the problem with tabs?
> 
> Yes, i added --app=$URL in order to open as on page APP

This is how it's supposed to look like? Because it's not loading my page. There is still only the unicorn logo

[image]

flivolsi76, 2022-12-20:
> > > Have you solved the problem with tabs?
> > 
> > 
> > Yes, i added --app=$URL in order to open as on page APP
> 
> This is how it's supposed to look like? Because it's not loading my page. There is still only the unicorn logo
> 
> [image]

Exactly, tell me if you resolved with that change as well!

majdzik84, 2022-12-20:
> Exactly, tell me if you resolved with that change as well!

Thank you. I re-uploaded the image and after your change it no longer shows tabs! (I think I messed something up earlier)