jareware/chilipie-kiosk #179: Discussion: singletonlock

sborn, 2022-03-29:
Occasionally our screens are frozen on reboot because of the singletonlock issue with chromium. Is there any reason not to run a shell script to delete it (if it exists) on each boot?

jareware, 2022-03-29:
Have not run into that, but certainly sounds like a problem.

What's this "singleton lock"? Sounds like it should be safe to delete before starting Chromium. 👍

sborn, 2022-03-29:
SingletonLock is a file in ~/.config/chromium/ that is created to identify the profile using Chromium. If it is still there on a fresh launch (because it terminates incorrectly or something; I can't figure out the variable), it will prevent Chromium from booting without a dialog box that asks if you're sure (because it would overwrite the other profile). It's annoying. I have a little shell script that deletes it now that is in a crontab on startup. It is even safe to delete it while the browser is up.

`rm /home/pi/.config/chromium/SingletonLock -f`

Here is an old link to where I figured out the fix. Note that the path to chromium is different now:
https://askubuntu.com/questions/297397/google-chrome-profile-in-use-browser-does-not-start-anymore

sborn, 2022-03-30:
Ahh, I figured it out. If you change the hostname and reboot it causes the locked Chromium. So this script eliminates that problem so that you can change the hostname headless and not worry about it.

jareware, 2022-04-04:
Oh wow, what an odd edge case. 😳 

Wanna make a PR with that in the default crontab? 😇