jareware/chilipie-kiosk #183: F11 not work

iadude10, 2022-04-11:
V 3.0.0-rc4
Rpi 3B+

F11 does nothing. Saw one other mention of this issue in discussion but no solution.
Could resolution be too high?

kleini, 2022-04-13:
I have the same issue but was able to work around it by right clicking and the context menu has an entry to exit fullscreen.

jareware, 2022-04-26:
Yes, we've seen this happen with the latest image. If someone is able to figure out a fix, I'll be happy to include it in the next build.

skater59, 2022-10-01:
I have the same problem with F11 and ctrl +L, right mouse klick doesnt work, as well I found a workaround using F1 for helpscreen, then using shortcut upper right corner to open new google chrome screen tab, then move mouse in the middle of the upper screen and close with klick on X button. Now fullscreen is terminated and adressline is here.