jareware/chilipie-kiosk #187: Clear cache and refresh on start?

sam98brown, 2022-07-22:
Is it possible to clear cache and refresh on boot? I have an issue where the page wont load without a cache clear and refresh.

sborn, 2022-07-22:
I have all of mine write-protected (overlay) so that they don’t get corrupted when my staff unplug them without shutting them down. That would serve the same function.

sam98brown, 2022-07-22:
Please could you point me in the right direction for doing that?

sam98brown, 2022-07-22:
Great, thanks for the help. Just so I'm sure, how does this ensure the cache issue won't be a problem?

sborn, 2022-07-22:
Have you ever booted from a “live CD” with anything? Where you don’t have a hard drive running the show, but the computer acts as if it is writing to a hard drive? Even though CDs can’t be written to?

sborn, 2022-07-22:
Anyway, it basically makes it like that. So every time it reboots, it is fresh again. Nothing that happens during previous sessions is stored. So before you enable all of that, make sure it is in the state that you want it to boot to every time.