jareware/chilipie-kiosk #200: Alternatives?

ColinTravis, 2023-12-19:
Does anybody know a decent alternative that possibly has been updated a little more recent? This has been my go-to kiosk image for years, but lately, any of my running kiosks after a few days of running and rebooting will just display a black screen and need to be re-flashed. I've tried a ton of fixes, and can't seem to crack it.

edbat, 2024-01-08:
I'd be interested in this too

karimd, 2024-02-05:
@ColinTravis @edbat you can start Chromium fullscreen from the command line by executing:

`chromium-browser --app=http://example.com --start-fullscreen`

BoKKeR, 2024-03-11:
To be able to start chrome you have to add this to your script 

`export DISPLAY=:0 && chromium-browser --app=http://my-site.com --start-fullscreen`